oneapi-src/oneAPI-samples #2084: C++ SYCL_FPGA GZIP example can't handle bigger input files

umairsiddiqui-digitek, 2023-12-06:
I compiled the C++ SYCL_FPGA GZIP example with default settings (2 Engines). seems like it GZIP example can't able to handle files bigger than 600MB. When i give bigger file application crashes

https://github.com/oneapi-src/oneAPI-samples/blob/277635f100309a1f1b062a21722786de29bf12a1/DirectProgramming/C%2B%2BSYCL_FPGA/ReferenceDesigns/gzip/src/gzip.cpp#L27

```
$ stat -c %s tmp_blob
536871802
$ ./gzip.fpga tmp_blob -o=tmp_blob.gz
Running on device: ofs_n6001 :  Intel Acceleration Development Platform N6001  (ofs_ed00000) 
Launching High-Bandwidth DMA GZIP application with 2 engines
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
outputSize: 536872058 Prepin: 0
kMinBufferSize: 16384 isz: 536871802 kInOutPadding: 256
Throughput: 6.69636 GB/s

TP breakdown for engine #0 (GB/s)
CRC = 5.69911
LZ77 = 3.35873
Huffman Encoding = 3.35535
DMA host-to-device = 8.31162
DMA device-to-host = 6.37311

TP breakdown for engine #1 (GB/s)
CRC = 5.91431
LZ77 = 3.36041
Huffman Encoding = 3.35313
DMA host-to-device = 9.14207
DMA device-to-host = 8.82735

Compression Ratio 22.7548%
PASSED
```

but when i provide bigger file 

```
$ stat -c %s linux-socfpga.tar
4339732480

$ ./gzip.fpga ./linux-socfpga.tar -o=linux-socfpga.tar.gz
Running on device: ofs_n6001 : Intel Acceleration Development Platform N6001 (ofs_ed00000)
Launching High-Bandwidth DMA GZIP application with 2 engines
outputSize: 44765440 Prepin: 0
kMinBufferSize: 16384 isz: 4339732480 kInOutPadding: 256
outputSize: 44765440 Prepin: 0
kMinBufferSize: 16384 isz: 4339732480 kInOutPadding: 256
Caught a SYCL host exception:
Native API failed. Native API returns: -4 (PI_ERROR_MEM_OBJECT_ALLOCATION_FAILURE) -4 (PI_ERROR_MEM_OBJECT_ALLOCATION_FAILURE)
terminate called after throwing an instance of 'sycl::_V1::runtime_error'
  what():  Native API failed. Native API returns: -4 (PI_ERROR_MEM_OBJECT_ALLOCATION_FAILURE) -4 (PI_ERROR_MEM_OBJECT_ALLOCATION_FAILURE)
Aborted (core dumped)
```

my host memory =>
```
$ free
              total        used        free      shared  buff/cache   available
Mem:       65583976     4882756    25815904       73340    34885316    59898552
Swap:       8138748      144640     7994108
```
```
$ grep -i hugepages /proc/meminfo
AnonHugePages:   1089536 kB
ShmemHugePages:    22528 kB
FileHugePages:   1548288 kB
HugePages_Total:    1024
HugePages_Free:     1024
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
```

KevinUTAT, 2024-05-02:
Hello, this design uses a large amount of DDR. And the crash is due to it's running out of DDR. This is a know limitation for this design.

yuguen-intel, 2024-11-25:
This was fixed in https://github.com/oneapi-src/oneAPI-samples/commit/8540c3679ca9bd5bdb8d068e4615bc13e4e84229

closed 2024-11-25 by yuguen-intel